foxglove/mcap #861: Cannot Access Message's Schema name and type to Create Schema with mcap ROS2

howird, 2023-03-17:
**Description**
The [MCAP Working with ROS2 Messages page](https://mcap.dev/docs/python/ros2_noenv_example.html) mentions that in a ROS2 environment, we can access the schema name using message._type and the schema text with message.__class__._full_text.
However, I cannot seem to access those fields.

- Version: mcap==1.0.2, mcap-ros2-support==0.3.0
- Platform: Ubuntu 22.04, ROS2 Humble

**Steps To Reproduce**
```
>>> from std_msgs.msg import Int64
>>> i = Int64()
>>> i.data = 5
>>> i._type
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Int64' object has no attribute '_type'
>>> i.__class__._full_text
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: type object 'Int64' has no attribute '_full_text'
```

**Expected Behavior**
No error

jtbandes, 2023-03-17:
Thanks for pointing this out, it looks like the docs may be wrong (I believe `_full_text` is a ROS 1 concept only). It's recommended to use rosbag2_py for working with ROS 2 mcap files, as described here: https://mcap.dev/docs/python/ros2_example.html

james-rms, 2023-03-30:
@howird I've made a PR to add this to `mcap-ros2-support`, but this won't work for your example code because you appear to be importing ROS 2 generated types, which we have no control over. After https://github.com/foxglove/mcap/pull/872 merges, you can use the `mcap-ros2-support` to decode ROS 2 MCAPs, and you will get objects with these attributes as expected.

defunctzombie, 2023-04-21:
Closing this issue since the referenced PRs merged.

closed 2023-04-21 by defunctzombie