foxglove/mcap #862: MCAP cli produces empty message index

mrkbac, 2023-03-17:
**Description**
The MCAP cli produces an empty message index instead of omitting it. It seems that this slows down Foxglove and the Python lib.

- Version: CLI v0.0.24
- Platform: mac OS

**Steps To Reproduce**
`mcap compress` on any file

**Expected Behavior**
no message index at all for topics not in this chunk

jtbandes, 2023-03-17:
Hi, thanks for the bug report, however, the "steps to reproduce" section is lacking. Can you provide more information about how one can see that this issue has occurred? And can you provide more detail about what slowdown you have observed?

mrkbac, 2023-03-18:
Sorry for the unclear explanation, for example the slow down occures in the following code:
```python
with open(sys.argv[1], "rb") as f:
    reader = make_reader(f)
    for schema, channel, message in reader.iter_messages(topics=["/tf_static"]):
        print(f"{channel.topic} ({schema.name}): {message.data}")
```

even after printing all `/tf_static` messages the reader keeps reading the whole file.
This only happens when the file was previously processed by the mcap cli (compress, filter, recover, ...), it does not happen for example when the mcap was produced by `ros2 bag record`
I also noticed slowdowns in foxglove studio when skipping in large files.

I've checked the record Length of the MessageIndex here https://github.com/foxglove/mcap/blob/0842cbf26dbf114a0fa0eb2dc36355f6db75382f/python/mcap/mcap/reader.py#L84-L86
which showed that a MessageIndex for `/tf_static` exists for all chunks but the record length is almost always zero.

jtbandes, 2023-03-18:
The code snippet is helpful, can you share an example file as well?

commit 741841ba (2023-03-30): go/mcap: do not write empty message indexes (#873)

### Public-Facing Changes

Go MCAP writer no longer produces message index records for channels not present in a chunk. This helps indexed readers skip chunks that do not contain any messages on a given channel.

### Description
Fixes #862
closed 2023-03-30 by james-rms