foxglove/mcap #888: mcap convert fails with "failed to convert file: unexpected EOF"

iliabaranov, 2023-05-10:
**Description**
mcap convert fails with "failed to convert file: unexpected EOF", generates a 1MB output message, when input bag file is ~300MB

- Version:
- Platform:

**Steps To Reproduce**

Installed using:
```wget https://github.com/foxglove/mcap/releases/latest/download/mcap-linux-amd64 -O mcap
chmod +x mcap```

ran with:
```
./mcap convert input.bag output.mcap
```
The input bagfile is not corrupt, reads fine in ROS1

**Expected Behavior**
Should be an mcap file?

jtbandes, 2023-05-10:
Can you provide a sample bag file that reproduces the problem?

iliabaranov, 2023-05-10:
Sure thing, I just sent this file as a DM to @defunctzombie on an unrelated lag topic with Foxglove

iliabaranov, 2023-05-10:
Ah, hang on, it might be because the rosbag contains compressed data.

Running same command on uncompressed data is much healthier.

Perhaps this bug should be changed to clarify that compressed data files are not supported?

james-rms, 2023-05-10:
Compressed `.bag` files are supported, though there may be an edge case here which we're not handling. I'll take a look and see what I can find. I've received the sample data you sent to @defunctzombie , will take a look.

iliabaranov, 2023-05-10:
Chatted with @defunctzombie, perhaps it's because we use bz2?
BZ2 does seem to be the default ROS1 compress config:
http://wiki.ros.org/rosbag/Commandline#compress

wkalt, 2023-05-10:
this does seem bz2 related somehow. That should be supported but I expect we encounter it less frequently than lz4. If I run the bag through rosbag compress like `rosbag compress obstacle_avoidance_2023-05-04-19-46-29_0.bag --lz4` the bag that comes out may then be converted.

Definitely a bug in here somewhere; the bag is fine.

wkalt, 2023-05-15:
@iliabaranov this is fixed in the latest CLI tool release, v0.0.31. Thanks for the report.

closed 2023-05-15 by wkalt

iliabaranov, 2023-05-15:
Thanks! :D