foxglove/mcap #895: mcap CLI cat_test fails

therishidesai, 2023-05-19:
When I run `go test ./...`  or `make test` in in `go/cli/mcap` to run the tests for the mcap CLI I get the following error:

```
   	github.com/foxglove/mcap/go/cli/mcap	[no test files]
--- FAIL: TestCat (0.00s)
    --- FAIL: TestCat/OneMessage (0.00s)
        cat_test.go:34:
            	Error Trace:	/home/rishi/mcap/go/cli/mcap/cmd/cat_test.go:34
            	Error:      	Expected nil, but got: &mcap.ErrBadMagic{actual:[]uint8{0x76, 0x65, 0x72, 0x73, 0x69, 0x6f, 0x6e, 0x20}}
            	Test:       	TestCat/OneMessage
panic: runtime error: invalid memory address or nil pointer dereference
	panic: runtime error: invalid memory address or nil pointer dereference [recovered]
	panic: runtime error: invalid memory address or nil pointer dereference
```

- Version: 0.0.31
- Platform: Linux

**Steps To Reproduce**

- `cd go/cli/mcap`
- `go test ./...`

**Expected Behavior**
 Tests should pass

jtbandes, 2023-05-19:
[image]

You are seeing a Git LFS pointer file. Git LFS needs to be installed and LFS-stored files downloaded before the test can work.

closed 2023-05-19 by jtbandes

joe-saronic, 2026-09-06:
@therishidesai Addressing this in nixpkgs via passthru: https://github.com/NixOS/nixpkgs/pull/560711