foxglove/mcap #901: mcap rust bindings fail to build due to binrw errors

therishidesai, 2023-05-30:
When using the latest versions of binrw the mcap rust bindings fail to build with the following error:

```
121 |     fn read_le<T: BinRead>(&mut self) -> BinResult<T>
    |                   ^^^^^^^ required by this bound in `BinReaderExt::read_le`

error[E0277]: the trait bound `MessageHeader: binrw::BinRead` is not satisfied
   --> /home/rishi/.cargo/registry/src/index.crates.io-6f17d22bba15001f/mcap-0.6.0/src/read.rs:423:74
    |
423 |             let header: records::MessageHeader = Cursor::new(header_buf).read_le()?;
    |                                                                          ^^^^^^^ the trait `binrw::BinRead` is not implemented for `MessageHeader`
    |
    = help: the following other types implement trait `binrw::BinRead`:
              ()
              (b1, b2, b3, b4, b5, b6, b7, b8, b9, b10, b11, b12, b13, b14, b15, b16, b17, b18, b19, b20, b21, b22, b23, b24, b25, b26, b27, b28, b29, b30, b31, b32)
              (b10, b11, b12, b13, b14, b15, b16, b17, b18, b19, b20, b21, b22, b23, b24, b25, b26, b27, b28, b29, b30, b31, b32)
              (b11, b12, b13, b14, b15, b16, b17, b18, b19, b20, b21, b22, b23, b24, b25, b26, b27, b28, b29, b30, b31, b32)
              (b12, b13, b14, b15, b16, b17, b18, b19, b20, b21, b22, b23, b24, b25, b26, b27, b28, b29, b30, b31, b32)
              (b13, b14, b15, b16, b17, b18, b19, b20, b21, b22, b23, b24, b25, b26, b27, b28, b29, b30, b31, b32)
              (b14, b15, b16, b17, b18, b19, b20, b21, b22, b23, b24, b25, b26, b27, b28, b29, b30, b31, b32)
              (b15, b16, b17, b18, b19, b20, b21, b22, b23, b24, b25, b26, b27, b28, b29, b30, b31, b32)
            and 46 others
note: required by a bound in `read_le`
   --> /home/rishi/.cargo/registry/src/index.crates.io-6f17d22bba15001f/binrw-0.9.2/src/binread/mod.rs:121:19
    |
121 |     fn read_le<T: BinRead>(&mut self) -> BinResult<T>
    |                   ^^^^^^^ required by this bound in `BinReaderExt::read_le`

error[E0433]: failed to resolve: use of undeclared type `McapString`
   --> /home/rishi/.cargo/registry/src/index.crates.io-6f17d22bba15001f/mcap-0.6.0/src/records.rs:212:17
    |
212 |         let k = McapString::read_options(reader, ro, args)?;
    |                 ^^^^^^^^^^ use of undeclared type `McapString`

error[E0433]: failed to resolve: use of undeclared type `McapString`
   --> /home/rishi/.cargo/registry/src/index.crates.io-6f17d22bba15001f/mcap-0.6.0/src/records.rs:213:17
    |
213 |         let v = McapString::read_options(reader, ro, args)?;
    |                 ^^^^^^^^^^ use of undeclared type `McapString`
```

- Version:
- Platform:

**Steps To Reproduce**

**Expected Behavior**

therishidesai, 2023-05-30:
Here are some more error messages related to this:
```
error: custom attribute panicked
   --> src/records.rs:105:1
    |
105 | #[binrw]
    | ^^^^^^^^
    |
    = help: message: attempt to subtract with overflow

error: proc-macro derive panicked
   --> src/records.rs:169:39
    |
169 | #[derive(Debug, Clone, Eq, PartialEq, BinRead, BinWrite)]
    |                                       ^^^^^^^
    |
    = help: message: attempt to subtract with overflow
  ```
  
  It looks like the updates to the `syn` crate is causing binrw to panic

therishidesai, 2023-05-30:
bumping the `binrw` version to `0.11.2` fixes these errors but now I get the following errors:
```
error[E0412]: cannot find type `ReadOptions` in this scope
   --> src/records.rs:131:10
    |
131 |     ro: &ReadOptions,
    |          ^^^^^^^^^^^ not found in this scope

error[E0412]: cannot find type `WriteOptions` in this scope
   --> src/records.rs:151:12
    |
151 |     opts: &WriteOptions,
    |            ^^^^^^^^^^^^ not found in this scope
    |
help: consider importing this struct
error[E0107]: missing generics for associated type `binrw::BinRead::Args`
   --> src/records.rs:129:32
    |
129 | fn parse_vec<T: binrw::BinRead<Args = ()>, R: Read + Seek>(
    |                                ^^^^ expected 1 lifetime argument
    |
note: associated type defined here, with 1 lifetime parameter: `'a`
   --> /home/rishi/.cargo/registry/src/index.crates.io-6f17d22bba15001f/binrw-0.11.2/src/binread/mod.rs:49:10
    |
49  |     type Args<'a>;
    |          ^^^^ --
help: add missing lifetime argument
    |
129 | fn parse_vec<T: binrw::BinRead<Args<'a> = ()>, R: Read + Seek>(
```

james-rms, 2023-05-30:
I don't think there's a complete set of steps to reproduce here, at least when I run
```
cd mcap/rust
cargo run --example mcapcat
```
on the latest main I don't see any build errors. we also run full builds and tests in CI on every commit, so i would expect to have caught this issue there.

`Cargo.toml` specifies binrw = "0.9", which is what I would expect to work here. Have you modified cargo.toml in your source tree?

therishidesai, 2023-05-30:
These errors are happening with the nightly toolchain. Going to change this issue into a feature request to bump `binrw` to `0.11`

defunctzombie, 2024-06-03:
We'd welcome a PR for this and some testing showing it is working but don't have the capacity to get to this ourselves right now.

james-rms, 2024-06-04:
Binrw is at `0.12.0` since mcap v0.9.0.

closed 2024-06-04 by james-rms