foxglove/mcap #926: Message.sequence: wording mistake?

omegacoleman, 2023-07-19:
**Description**

Hi, I'm trying to implement a mcap-compatible bag facility for a self-driving system. While I appreciate your hard work, I suspect a wording mistake in `spec/index.md`:

```markdown
### Message (op=0x05)

...

| 4     | sequence     | uint32    | Optional message counter assigned by publisher. If not assigned by publisher, must be recorded by the recorder. |
```

It says the sequence is optional, but "If **not** assigned by publisher, must be recorded by the recorder.", wouldn't that contradict with 'optional'?

It seems like a wording mistake of "If assigned by publisher, must be recorded by the recorder."

omegacoleman, 2023-07-19:
I proposed a fix in #927

jtbandes, 2023-07-20:
Hi, I believe this is not really a wording mistake per se... it's trying to say that since the field is required, if the publisher does not provide a sequence number then the recorder will have to come up with one. However, I agree that the wording is unclear. We will discuss internally.

omegacoleman, 2023-07-20:
> Hi, I believe this is not really a wording mistake per se... it's trying to say that since the field is required, if the publisher does not provide a sequence number then the recorder will have to come up with one.

Thanks for clarifying that out, this is important to me.

> However, I agree that the wording is unclear. We will discuss internally.

I'll leave this open for a while, since that would still contradict with the `Optional` part.

My suggestions would be:
- if the recorder `MUST` come up with a sequence, then the field is not optional, it's mandatory
- Add the `the recorder will have to come up with one` part to the specs to let everybody know

wkalt, 2023-07-28:
@omegacoleman I have raised a PR here to propose some new wording if you're interested: https://github.com/foxglove/mcap/pull/938

PR #938 (2023-07-31): Spec: improve description of sequence field
Fixes: #926
commit 75e3cab3 (2023-07-31): Spec: improve description of sequence field (#938)

Fixes: #926
closed 2023-07-31 by wkalt