foxglove/mcap #939: mcap cli merge does not copy metadata

mckay6842, 2023-08-01:
**Description**
When merging two mcap files via the cli, the output mcap file does not contain metadata records that were present in the input files.

- Version: go/mcap/v1.0.1
- Platform: Linux

**Steps To Reproduce**
```bash
# starting with two mcap files test.mcap and test2.mcap (both containing no metadata)

> mcap info test.mcap | grep metadata
metadata: 0
> mcap info test2.mcap | grep metadata
metadata: 0

# add metadata to one of the files
> mcap add metadata --name "test" --key "key=value" test.mcap

# check if metadata is now present
> mcap info test.mcap | grep metadata
metadata: 1
> mcap get metadata --name "test" test.mcap                  
{
  "key": "value"
}

# merge the two files
> mcap merge test.mcap test2.mcap -o merged.mcap

# check the output file
> mcap info merged.mcap | grep metadata                                                                                                                                  1 
metadata: 0
> mcap get metadata --name "test" merged.mcap 
failed to fetch metadata: metadata test does not exist

```

**Expected Behavior**
metadata (and attachments) should be copied as well

achim-k, 2023-09-08:
Attachments are also not copied, adapted the title

defunctzombie, 2023-12-04:
Metadata is now copied. Attachments are being worked on: https://linear.app/foxglove/issue/FG-5548/attachments-are-lost-when-using-mcap-merge

defunctzombie, 2024-02-20:
Attachments should also be included now.