foxglove/mcap #941: conan 2.0 support

l3enQ, 2023-08-03:
**Description**
the current recipe is not supported in conan 2.0 and cannot built with it.
the dependencies (lz4 and zstd) are updated for conan 2.0 so it theoritically should be built using the header file and adding dependencies to conan instead of mcap itself

- Version: all versions
- Platform: all platforms

**Steps To Reproduce**
from conans import ConanFile, CMake

class McapExamplesConan(ConanFile):
    settings = "os", "compiler", "build_type", "arch"
    generators = "cmake"
    requires = [
        "mcap/1.2.0",
        "protobuf/3.21.1",
        "nlohmann_json/3.10.5",
        "catch2/2.13.8",
    ]

    def build(self):
        cmake = CMake(self)
        cmake.configure()
        cmake.build()

**Expected Behavior**
[requires]
mcap/1.2.0

[generators]
cmake

defunctzombie, 2023-08-03:
@l3enQ Can you suggest a fix for the conanfile: https://github.com/foxglove/mcap/blob/ec08e50585355cac380e7fb895498871f916b408/cpp/mcap/conanfile.py#L4 since you are familiar with conan 2.0.

l3enQ, 2023-08-04:
using the header in the project and adding dependencies in `conanfile.txt` make it work in conan  2.0
the procedures exist in conan website.
e.g., https://conan.io/center/recipes/zstd?version=1.5.5 for zstd
but the one for mcap is empty https://conan.io/center/recipes/mcap?version=1.2.0

[requires]
zstd/1.5.2
lz4/1.9.3

[generators]
CMakeDeps
CMakeToolchain

defunctzombie, 2023-08-04:
From what I can tell of the package in conan center: https://github.com/conan-io/conan-center-index/tree/master/recipes/mcap/all there is already some v2 support via: https://github.com/conan-io/conan-center-index/pull/15649

But I also see:
https://github.com/foxglove/mcap/issues/609

If you know what should be fixed in this repo or the conan-center repo to resolve the issues you are seeing with v2 we'd be happy to look at a PR that fixes these.

l3enQ, 2023-08-04:
Actually, I don't know much about how conan should work on your side!
but it seems the conan-center bot couldn't build the packag; so nothing is uploaded! maybe something should be updated on the https://github.com/conan-io/conan-center-index/tree/master/recipes/mcap/

jtbandes, 2023-08-15:
> it seems the conan-center bot couldn't build the packag; so nothing is uploaded!

Can you please try again? I just visited [this page](https://conan.io/center/recipes/mcap?version=1.2.0) and I see that the latest versions are available: 
[image]

We added these versions in https://github.com/conan-io/conan-center-index/pull/18784

l3enQ, 2023-08-16:
looks like it is updated in the conan center
I will test and if this works, I will close the issue

l3enQ, 2023-08-17:
it builds ok now with conan 2.0 but the recipe is still poor on the conan center compared to the others like zstd
special as the c++ section of the repo is still based on conan 1.6.0

closed 2023-08-17 by l3enQ

l3enQ, 2023-08-18:
there is sime "Useful information to take into account to consume this library" in zstd and lz4 which we can see the tarets and the headers for usae but there is nothin for mcap

jtbandes, 2023-08-18:
It seems like that info might be from the `package_info()` in the conanfile: https://github.com/conan-io/conan-center-index/blob/master/recipes/zstd/all/conanfile.py#L81

You could make a PR to conan-center-index to improve it!