foxglove/mcap #943: Mcap Decoder is not Capable of Decoding "std_msgs/msg/Empty"

shouhengyi, 2023-08-08:
**Description**
`mcap` decoder can't handle "std_msgs/msg/Empty" type and will throw exception.

- Version: `mcap==1.1.0`
- Platform: This was on ros-iron image with the `mcap==1.1.0` and `mcap-ros2-support==0.5.0` installed (both the latest releases)

**Steps To Reproduce**
1. Record a bag containing a "std_msgs/msg/Empty" message in MCAP format; either by running `ros2 bag record -a -s mcap -o empty_message_bag` and `ros2 topic pub empty_msg std_msgs/msg/Empty`, or by using Write Messages example in https://mcap.dev/docs/python/ros2_example#writing-messages.
2. Attempt to read all messages using the official Reading Messages example in https://mcap.dev/docs/python/ros2_noenv_example#reading-messages

**Expected Behavior**
mcap decoder can decode "std_msgs/msg/Empty" in the same way that the official rosbag2_py SequentialReader can do.

**Actual Behavior**
mcap decoder raises an exception when reading from a bag that contains a "std_msgs/msg/Empty" message.

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/ubuntu/robolog/ros2/ros2_bag_reader.py", line 274, in messages
    message = next(self._messages)
  File "/root/ubuntu/.venv/lib/python3.10/site-packages/mcap/reader.py", line 194, in iter_decoded_messages
    schema, channel, message, decoded_message(schema, channel, message)
  File "/root/ubuntu/.venv/lib/python3.10/site-packages/mcap/reader.py", line 182, in decoded_message
    decoder = factory.decoder_for(channel.message_encoding, schema)
  File "/root/ubuntu/.venv/lib/python3.10/site-packages/mcap_ros2/decoder.py", line 45, in decoder_for
    raise McapROS2DecodeError(f'schema parsing failed for "{schema.name}"')
mcap_ros2.decoder.McapROS2DecodeError: schema parsing failed for "std_msgs/msg/Empty"
```

shouhengyi, 2023-08-15:
Ping on this. Thanks.

shouhengyi, 2023-08-21:
Ping on this again. Thanks.

closed 2023-08-23 by achim-k

achim-k, 2023-08-23:
This is fixed as of [version 0.5.1](https://pypi.org/project/mcap-ros2-support/0.5.1/)

PR #949 (2023-08-23): mcap_ros2: Fix serde of empty messages
### Public-Facing Changes

mcap_ros2: Fix serde of empty messages

### Description
In case a message definition definition is empty, ROS 2 adds a `uint8 structure_needs_at_least_one_member` field when converting to IDL, to satisfy the requirement from IDL of not being empty. See also
https://design.ros2.org/articles/legacy_interface_definition.html

This patch inserts or skips that extra member when serializing or deserializing respectively.

This PR is pretty much equivalent to https://github.com/foxglove/rosmsg2-serialization/pull/16

Fixes #943
commit 2a1f9c62 (2023-08-23): mcap_ros2: Fix serde of empty messages (#949)

### Public-Facing Changes

mcap_ros2: Fix serde of empty messages

### Description
In case a message definition definition is empty, ROS 2 adds a `uint8
structure_needs_at_least_one_member` field when converting to IDL, to
satisfy the requirement from IDL of not being empty. See also
https://design.ros2.org/articles/legacy_interface_definition.html

This patch inserts or skips that extra member when serializing or
deserializing respectively.

This PR is pretty much equivalent to
https://github.com/foxglove/rosmsg2-serialization/pull/16

Fixes #943