foxglove/mcap #976: C++: Writen metadata overlaps with writen attachment

juanlofer-eprosima, 2023-09-19:
**Description**

We use MCAP C++ library in our application to write and read a serialized object in the attachments section.
All seemed to work nicely until we started to write a metadata map in addition to the attachment. After this, deserialization of the attachment fails, and while debugging, we could observe bytes corresponding to the metadata in the attachment's data.
Inverting the writing order seems to be sufficient to circumvent the problem, although it would be great to remove this limitation.

**Note:**
We introduced a slight change into the library, adding two new attributes and getters for metadatas and attachments in the ``McapReader`` class, so that it is possible to fetch those easily. However, we do not believe this might have anything to do with the issue.

- Version: [releases/cpp/v1.1.0](https://github.com/foxglove/mcap/releases/tag/releases%2Fcpp%2Fv1.1.0)
- Platform: Ubuntu Focal

**Steps To Reproduce**

Write an attachment, then a metadata map. Attachment's data is corrupted and can no longer be parsed.

**Expected Behavior**
Writing metadata should not interfere with previously written attachments.

jtbandes, 2023-09-19:
Could you provide some sample code that demonstrates the issue?

juanlofer-eprosima, 2023-09-20:
[mcap-reproducer.zip](https://github.com/foxglove/mcap/files/12671274/mcap-reproducer.zip)

Here you can find two simple scripts with which I could reproduce the issue. These are a simplification of what is being done in our [application](https://github.com/eProsima/DDS-Record-Replay). In case you prefer to directly build and look at our application, make sure to build [this version](https://github.com/eProsima/DDS-Record-Replay/tree/9e1ac1e6071390dff2839d968678b675b9d60ccc) (before a workaround to avoid this issue was introduced).

I also retested after updating mcap to the [latest version](https://github.com/foxglove/mcap/tree/0a06331980a47ef606cca06b6ee3fc987f5f2d52) available in main branch. You can check the only differences I applied to the library by looking at the patch within the zip.

Best regards

james-rms, 2023-11-17:
Hi,

I ran the following (modified) version of your reproduction:
```cpp
#define MCAP_IMPLEMENTATION
#include <mcap/mcap.hpp>

int main()
{
    std::string filename = "reproducer.mcap";

    // Create writer and open file
    mcap::McapWriter mcap_writer;
    mcap::McapWriterOptions writer_options{"ros2"};
    auto status = mcap_writer.open(filename.c_str(), writer_options);

    // Write attachment
    std::string attachment_str = "AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA";
    std::cout << "Attachment string: " << attachment_str << std::endl;
    mcap::Attachment attachment;
    attachment.name = "my_attachment";
    attachment.data = (std::byte*)reinterpret_cast<const unsigned char*>(attachment_str.c_str());
    attachment.dataSize = attachment_str.size();
    status = mcap_writer.write(attachment);

    // Write metadata AFTER attachment
    mcap::KeyValueMap version;
    version["release"] = "v1.2.3";
    version["commit"] = "whatever";
    mcap::Metadata version_metadata;
    version_metadata.name = "version";
    version_metadata.metadata = version;
    status = mcap_writer.write(version_metadata);

    // Close file
    mcap_writer.close();

    return 0;
}
```

And it produces the following file: 
[reproducer.mcap.zip](https://github.com/foxglove/mcap/files/13385514/reproducer.mcap.zip)

I can read out the attachment and the metadata using the `mcap CLI`:
```
j@192-168-1-108 cpp % mcap get attachment reproducer.mcap --name my_attachment | cat
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA%                                                                     
j@192-168-1-108 cpp % mcap get metadata reproducer.mcap --name version              
{
  "commit": "whatever",
  "release": "v1.2.3"
}
j@192-168-1-108 cpp % 
```

so i suspect the writer is not the issue.

I do notice an issue with your patch to the reader code.
You store a multimap of attachment names to `Attachment` structs, but Attachments _do not own the allocation for their data_, and the data pointer will be invalidated after the next read. You cannot safely store `Attachment`s without copying from the `data` pointer to a separate buffer.

closed 2023-11-17 by james-rms

juanlofer-eprosima, 2023-11-17:
Thanks @james-rms for looking into this, I will modify the reader so it copies internal data. Thanks again!