foxglove/mcap #997: "unknown field SkipMagic" when installing cli mcap from github url, works fine in cloned source

lucasw, 2023-10-24:
**Description**

- Version: 8bc1ef7ae4eb949bdcce8a8a61f1a036819847fc  tag: releases/python/mcap-ros2-support/v0.5.3
- Platform: Ubuntu 23.04, apt installed golang-go `go1.20.3 linux/amd64`

**Steps To Reproduce**

```
sudo apt install golang-go
$ go install github.com/foxglove/mcap/go/cli/mcap@latest
# github.com/foxglove/mcap/go/cli/mcap/utils
go/pkg/mod/github.com/foxglove/mcap/go/cli/mcap@v0.0.0-20231023205342-8bc1ef7ae4eb/utils/mcap_amendment.go:189:3: unknown field SkipMagic in struct literal of type mcap.WriterOptions
go/pkg/mod/github.com/foxglove/mcap/go/cli/mcap@v0.0.0-20231023205342-8bc1ef7ae4eb/utils/mcap_amendment.go:259:17: writer.WriteChunkIndex undefined (type *mcap.Writer has no field or method WriteChunkIndex)
go/pkg/mod/github.com/foxglove/mcap/go/cli/mcap@v0.0.0-20231023205342-8bc1ef7ae4eb/utils/mcap_amendment.go:401:3: unknown field SkipMagic in struct literal of type mcap.WriterOptions
```

Cloning locally works:

```
git clone git@github.com:foxglove/mcap.git  # probably should do shallow clone instead
go install go/cli/mcap
ls ~/go/bin
$HOME/go/bin/mcap
🔪 Officially the top-rated CLI tool for slicing and dicing MCAP files.
...
```

(`mcap version` just outputs a blank line though)

**Expected Behavior**

`mcap` binary is put into `$HOME/go/bin`

jtbandes, 2023-10-24:
I think that’s why we say “Installing via 'go install' is not recommended, and is not guaranteed to work.” 😉

jtbandes, 2023-10-24:
Some context: https://github.com/foxglove/mcap/pull/826

lucasw, 2023-10-24:
Ok, it looks like from https://github.com/foxglove/mcap/blob/main/.github/workflows/ci.yml `make` is the real way to build it:

```
cd go
make -C cli/mcap build
...
./cli/mcap/bin/mcap version
releases/python/mcap-ros2-support/v0.5.3
```

PR #1022 (2023-11-21): Remove erroneous `go install` instructions
The existing instructions say that `go install` is not recommended and then proceed to give you a command that we know does not work. This change removes the incorrect command and provides a set of instructions for building the CLI from source.

Fixes: #997
commit a44c048f (2023-11-21): Remove erroneous `go install` instructions (#1022)

The existing instructions say that `go install` is not recommended and
then proceed to give you a command that we know does not work. This
change removes the incorrect command and provides a set of instructions
for building the CLI from source.

Fixes: #997

---------
closed 2023-11-21 by jtbandes