foxglove/mcap #1002: Missing documentation for `offered_qos_profiles`

jhurliman, 2023-11-01:
**Description**

The ROS 2 section of MCAP's well-known profiles documentation (https://mcap.dev/spec/registry#ros2) says `offered_qos_profiles` is a required metadata field, but no documentation or example is provided for what this field should contain. Is it JSON-serialized? What are the fields?

jhurliman, 2023-11-01:
Related: https://github.com/ros2/rosbag2/issues/1420

defunctzombie, 2023-11-07:
@MichaelOrlov @emersonknapp Reading the referenced rosbag issue its not clear to me how to document this or whether the mcap spec is even the correct place to document this.

Options that come to mind:

* Update the spec to say that the field is a string but the string is ROS distro/release specific?
* Remove the documentation of the field from the mcap spec altogether though arguably that makes creating tooling for ROS 2 mcap files harder?
* Be more specific and say that the string is valid yaml? (is it?)

I also noticed that the mcap storage plugin also provides a type hash field now: https://github.com/ros2/rosbag2/blob/rolling/rosbag2_storage_mcap/src/mcap_storage.cpp#L820 which did not exist when we first wrote this profile spec.

MichaelOrlov, 2023-11-09:
@defunctzombie @jhurliman Can we refer to the section about QoS profile overriding in the Rosbag2 readme file https://github.com/ros2/rosbag2#overriding-qos-profiles and official [How-to Overriding QoS Policies in rosbag2 guide](https://docs.ros.org/en/rolling/How-To-Guides/Overriding-QoS-Policies-For-Recording-And-Playback.html)?

jhurliman, 2023-11-09:
@MichaelOrlov the only tricky part about those docs is the required format is _slightly_ different. The top-level `/topic_name:` key cannot be present, only the key-values below that. That's what the rosbag2 recorder is writing and expecting to read, anyways. The following "schema" from https://docs.ros.org/en/rolling/How-To-Guides/Overriding-QoS-Policies-For-Recording-And-Playback.html is exactly the documentation that should be provided or linked to, somehow:

```
history: [keep_all, keep_last]
depth: int
reliability: [system_default, reliable, best_effort, unknown]
durability: [system_default, transient_local, volatile, unknown]
deadline:
  sec: int
  nsec: int
lifespan:
  sec: int
  nsec: int
liveliness: [system_default, automatic, manual_by_topic, unknown]
liveliness_lease_duration:
  sec: int
  nsec: int
avoid_ros_namespace_conventions: [true, false]
```

PR #1020 (2023-12-01): Add link to ROS QoS Policy docs to spec registry
Update the `ros2` profile section of the registry with more details about the expectations for `offered_qos_profiles` and link to the official ROS docs.

Fixes: #1002

---
I do wonder if it would be more appropriate for the ROS docs to define these aspects of the profile and the MCAP registry links to those docs for details on the profile.
commit d40dba52 (2024-10-30): Add link to ROS QoS Policy docs to spec registry (#1020)

Update the `ros2` profile section of the registry with more details
about the expectations for `offered_qos_profiles` and link to the
official ROS docs.

Fixes: #1002
closed 2023-12-01 by defunctzombie