foxglove/mcap #1007: Frequency calculation uses entire MCAP duration as the denominator instead of per-channel duration

jhurliman, 2023-11-01:
**Description**

If I have a 10 minute MCAP file containing two channels, where the first channel contains one message per second for 10 minutes and the second channel also contains one message per second but starting at the 5 minute mark, the frequencies will be reported as 1 Hz and 0.5 Hz.

The per-channel frequency calculation should, in my opinion, use the per-channel duration instead of the total MCAP duration.

james-rms, 2023-11-01:
The per-channel duration is not available without reading the entire file:

* The Statistics record only stores the all-channel start and end times
* Chunk records only store all-channel start and end times for that chunk

We can maybe revisit this as part of a major spec revision.

closed 2023-11-01 by jtbandes

defunctzombie, 2023-11-02:
I don't know that it would require a major spec revision - but it would require a new kind of summary record (or other new records) to avoid reading the entire file.

Possibly useful would be a disclaimer on the _info_ command that the _hz_ readings are estimates? Maybe that is too pedantic since the hz could vary throughout the data.