foxglove/mcap #1035: integrating with custom library

harshitsankhla, 2023-12-07:
**Description**

I am facing build issues when linking with my custom library.

- Platform: linux

**Steps To Reproduce**
A minimal setup would be based on this file structure

src - 
---mywriter.cpp (uses mcap::Mcapwriter)
---myreader.cpp (uses mcap::Mcapreader)
---common.cpp (uses mcap::schema) -> also where Mcap_Implementation is defined

include - corresponding .hpp files with the same name

example.cpp - creates mywriter object (include<mywriter.hpp> and accesses its methods)

I am able to link to the library but get  abort core dump errors when executing. Even if I keep the mywriter class empty with just an empty constructor and define only a mcap::writer variable in the class, it still fails. 

It works perfectly fine when everything (src + include) is just in a single file say mycode.h and then use it in example. I feel the mcap_implementation define might be causing issues?

Any tips?
    

**Expected Behavior**

harshitsankhla, 2023-12-11:
```
cmake_minimum_required(VERSION 3.14)
project(swiftlogging)

find_package(Protobuf REQUIRED)
message(STATUS "Using Protocol Buffers ${Protobuf_VERSION}")

# # Get all CMake properties
# get_cmake_property(all_variables VARIABLES)

# # Iterate over the variables and print their names and values
# foreach(variable ${all_variables})
#     message(STATUS "${variable} = ${${variable}}")
# endforeach()

set(CMAKE_BUILD_TYPE Debug)

set(LOGGING_INCLUDE_DIRS
    include_directories(${CMAKE_CURRENT_SOURCE_DIR}/include)
    include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../../thirdparty/mcap/include)
    include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../messages/generated)
    include_directories(${Protobuf_INCLUDE_DIRS})
)
# include_directories(${LOGGING_INCLUDE_DIRS})

add_definitions(-DMCAP_COMPRESSION_NO_ZSTD)
add_subdirectory("../messages" swiftmessages)

set(CMAKE_INTERPROCEDURAL_OPTIMIZATION TRUE)

# create library for core algorithms
add_library(${PROJECT_NAME}
    ${PROJECT_SOURCE_DIR}/src/common.cpp
    ${PROJECT_SOURCE_DIR}/src/swiftReader.cpp
    ${PROJECT_SOURCE_DIR}/src/swiftWriter.cpp
)
target_include_directories(${PROJECT_NAME} PUBLIC ${LOGGING_INCLUDE_DIRS})
target_link_libraries(${PROJECT_NAME} PUBLIC swiftmessaging ${Protobuf_LIBRARIES} lz4)
set_target_properties(${PROJECT_NAME} PROPERTIES CXX_STANDARD 17 CXX_STANDARD_REQUIRED ON)

# create executable examples
add_executable(example ${PROJECT_SOURCE_DIR}/examples/example.cpp)
target_include_directories(example PRIVATE ${CMAKE_CURRENT_SOURCE_DIR}/include)
target_link_libraries(example PRIVATE ${PROJECT_NAME})
set_target_properties(example PROPERTIES CXX_STANDARD 17 CXX_STANDARD_REQUIRED ON)
# target_link_libraries(example PUBLIC swiftmessaging protobuf::libprotobuf lz4)
```

james-rms, 2024-01-07:
I can't reproduce this.

closed 2024-01-07 by james-rms