foxglove/mcap #1057: data_stream: OverflowError: cannot fit 'int' into an index-sized integer

mxterry, 2024-01-26:
**Description**
when reading a series of mcap file with ros2 format, the error occurred:

`OverflowError: cannot fit 'int' into an index-sized integer`

- Version: 1.1.1
- Platform:  Ubuntu 22.04

**Steps To Reproduce**
Sorry that it seems the data is changed in some reason when I rewrite my script into single-process mode in order to debug the source data. 

I'm still find a way to locate the mcap data if it's possible and I will update later.

**Expected Behavior**
```python
Traceback (most recent call last):
  File "/home/users/chenguang.xue/dev/evalsdk/.venv/lib/python3.8/site-packages/mcap/reader.py", line 278, in iter_messages
    summary = self.get_summary()
  File "/home/users/meterry/dev/evalsdk/.venv/lib/python3.8/site-packages/mcap/reader.py", line 338, in get_summary
    footer = next(StreamReader(self._stream, skip_magic=True).records)
  File "/home/users/meterry/dev/evalsdk/.venv/lib/python3.8/site-packages/mcap/stream_reader.py", line 120, in records
    record = self._read_record(opcode, length)
  File "/home/users/meterry/dev/evalsdk/.venv/lib/python3.8/site-packages/mcap/stream_reader.py", line 205, in _read_record
    self._stream.read(length - 9)
  File "/home/users/meterry/dev/evalsdk/.venv/lib/python3.8/site-packages/mcap/data_stream.py", line 26, in read
    data = self._stream.read(length)
OverflowError: cannot fit 'int' into an index-sized integer
```

jtbandes, 2024-01-26:
It seems like the original mcap file is corrupt. if you cannot share the file then perhaps you can provide complete steps to reproduce the issue, or generate a sample file with non-private data which demonstrates the same issue.

defunctzombie, 2024-01-26:
> Theoretically, mock a such long number and assign to `length` during debugging, could lead to this OverflowError.

While that could reproduce the error, this does not actually lead to resolving the issue which is that the original mcap file had some bytes in a location interpreted as a _length_ of some record or message that decoded into this large number. As @jtbandes pointed out this usually means the file is corrupt in some way. If you cannot share the file there is not much else for the community or us to do on an issue like this.

If you can create reproduction steps (that involve a writer rather than manually crafting a broken length file) then we can investigate if there are issues in a writer. I am going to close this since to us it looks like a corrupt file. Happy to re-visit if you are able to do some digging on your end with your private setup and provide public steps.

closed 2024-01-26 by defunctzombie