foxglove/mcap #1063: Request from Homebrew maintainers

p-linnane, 2024-02-14:
Hello 👋 . I'm a maintainer for the [Homebrew](https://brew.sh) project. I'm reaching out regarding the automated version bump PR's for mcap. As part of our merge process, we push our built bottles to the PR branch. Since your automated PR's are being opened from an org fork, we are unable to push to it. This requires us to recreate a new PR to get the bottles merged. I'd like to request that you open future automated PR's from a non-org fork to simplify the merge process for us. Thanks!

amacneil, 2024-02-14:
Hey, thanks for flatting. I think https://github.com/foxglove/mcap/pull/1064 will fix this.

I also submitted a readme update https://github.com/mislav/bump-homebrew-formula-action/pull/150 to `mislav/bump-homebrew-formula-action` which will hopefully reduce the number of others who fall into the same trap.

amacneil, 2024-02-14:
We won't know for sure if this worked until we do a new release, but I can keep an eye on it.

closed 2024-02-14 by amacneil

p-linnane, 2024-04-24:
mcap has been added to our autobump list, which means we automatically check for new releases every few hours and automatically open a PR if one is found. You shouldn't need to worry about opening PR's.

jtbandes, 2024-04-24:
Is there any way the maintainers of this project can be notified if a future bump fails to build? Will the homebrew maintainers open an issue here?

p-linnane, 2024-04-24:
We will try to fix the build, and if we can't then we would open an issue here.