foxglove/mcap #1147: loadMessageIndexes() must be called before hasMore()

fredrikjacobson, 2024-03-29:
**Description**

Hi, when trying out mcap using latest version I receive an error that `loadMessageIndexes() must be called before hasMore()` from `ChunkCursor`. When using `2.0.2` the issue seems to disappear. 

- Version:  `@mcap/core": "^2.1.0`
- Platform: Typescript (both node and browser).

**Steps To Reproduce**

```typescript
const file = await fetch(
    "https://url-to-mcap/scene.mcap"
  );

const decompressHandlers = await loadDecompressHandlers();
const reader = await McapIndexedReader.Initialize({
  readable: new BlobReadable(await file.blob()),
  decompressHandlers,
});

for await (const message of reader.readMessages({
        topics: ['/CAM_FRONT_RIGHT/image_rect_compressed']
      })) {
    console.log(message.logTime);
}

```

**Expected Behavior**

Should parse message without errors.

jtbandes, 2024-03-29:
Thanks for the report. I believe this should be fixed by https://github.com/foxglove/mcap/pull/1149.

jtbandes, 2024-03-29:
Published v2.1.1

PR #1149 (2024-03-29): @mcap/core: Fix error when skipping chunks
### Public-Facing Changes

@mcap/core: fixed an error when reading topics that were only present in some of the file's chunks.

### Description

Fixes #1147, a regression which I believe was caused by #1037. Calling `loadMessageIndexes()` is supposed to guarantee that `#orderedMessageOffsets` is set.
commit bce280bb (2024-03-29): @mcap/core: Fix error when skipping chunks (#1149)

### Public-Facing Changes

@mcap/core: fixed an error when reading topics that were only present in
some of the file's chunks.

### Description

Fixes #1147, a regression which I believe was caused by #1037. Calling
`loadMessageIndexes()` is supposed to guarantee that
`#orderedMessageOffsets` is set.
closed 2024-03-29 by jtbandes