foxglove/mcap #1198: Even after reinstalling, the mcap tool is still failing with the same memory allocation error.

josephykwang, 2024-07-17:
**Description**

% mcap version
v0.0.47

MacBook Pro with Sonoma 14.5 (23F79)

**Steps To Reproduce**

% od -t x1 output.mcap| head -3
0000000    89  4d  43  41  50  30  0d  0a  00  00  00  43  7b  27  66  6f
0000020    72  6d  61  74  27  3a  20  27  68  32  36  34  27  2c  20  27
0000040    77  69  64  74  68  27  3a  20  31  39  33  36  2c  20  27  68
x
% mcap info output.mcap

Failed to read file output.mcap: failed to get reader: could not read MCAP header when opening reader: failed to allocate 8245921762496413696 bytes for reserved token: length out of int32 range

% mcap doctor output.mcap
Failed to read next token: failed to allocate 8245921762496413696 bytes for reserved token: length out of int32 range

here is the code that converting png files to h264 format and then to MCAP file. 

```
import os
import ffmpeg
import struct

def convert_png_to_h264(frames_dir, output_h264_file, frame_rate):
    frame_files = sorted([f for f in os.listdir(frames_dir) if f.endswith('.png')])
    
    if not frame_files:
        print("No PNG frames found in the directory.")
        return

    # Remove the existing output file if it exists
    if os.path.exists(output_h264_file):
        os.remove(output_h264_file)

    # Use ffmpeg to convert PNG frames to H.264
    input_files = os.path.join(frames_dir, 'frame_%04d.png')
    ffmpeg.input(input_files, framerate=frame_rate).output(output_h264_file, vcodec='libx264', pix_fmt='yuv420p').run()

def process_h264_to_mcap_custom(h264_file, mcap_file, width, height, frame_rate, chunk_size=1024*1024):  # 1MB chunks
    with open(h264_file, 'rb') as h264, open(mcap_file, 'wb') as f:
        # Write the MCAP magic header
        f.write(b"\x89MCAP0\r\n")
        print("Written MCAP header.")
        
        # Write the metadata
        metadata = {
            "format": "h264",
            "width": width,
            "height": height,
            "frame_rate": frame_rate,
        }
        metadata_bytes = bytes(str(metadata), 'utf-8')
        metadata_length = len(metadata_bytes)
        print(f"Metadata length: {metadata_length}")
        f.write(struct.pack('>I', metadata_length))  # Write metadata length as a 4-byte big-endian integer
        f.write(metadata_bytes)
        print(f"Written metadata: {metadata}")

        # Write the frame data in chunks
        total_written = 0
        while True:
            frame_chunk = h264.read(chunk_size)
            if not frame_chunk:
                break
            frame_data_length = len(frame_chunk)
            print(f"Frame chunk data length: {frame_data_length}")
            f.write(struct.pack('>I', frame_data_length))  # Write frame chunk length as a 4-byte big-endian integer
            f.write(frame_chunk)
            total_written += frame_data_length
            print(f"Total data written so far: {total_written}")
        
        print("Finished writing MCAP file")
        print(f"Final total data written: {total_written}")

def convert_frames_to_mcap(frames_dir, output_dir, output_h264_file, width=1920, height=1080, frame_rate=30):
    if not os.path.isdir(frames_dir):
        print(f"Directory {frames_dir} does not exist.")
        return

    # Convert PNG frames to H.264
    convert_png_to_h264(frames_dir, output_h264_file, frame_rate)

    # Process H.264 file into custom MCAP
    mcap_file = os.path.join(output_dir, 'output.mcap')
    process_h264_to_mcap_custom(output_h264_file, mcap_file, width, height, frame_rate)

    print(f"Generated MCAP file: {mcap_file}")

# Create the output directory if it does not exist
output_dir = '/mnt/data/mcap_files'
os.makedirs(output_dir, exist_ok=True)

# Define the output H.264 file
output_h264_file = '/mnt/data/output_frames/output.h264'

# Convert the frames to MCAP files
convert_frames_to_mcap('/mnt/data/output_frames', output_dir, output_h264_file, width=1936, height=1220, frame_rate=30)
```

**Expected Behavior**
gemini suggested it is a bug

```
Even after reinstalling, the mcap tool is still failing with the same memory allocation error. This suggests a few remaining possibilities:

Very Specific Bug: There might be a rare, corner-case bug in the mcap tool that only triggers with certain file configurations. The developers might not be aware of it yet.

System-Level Issue: It's possible (though less likely) that there's something unique about your macOS setup or environment that's causing the tool to malfunction.

File Size and Limitations: While we confirmed the header looks valid, the sheer size of the file (500MB) could still be pushing the limits of the tool or the system's resources.

What to Try Next:

Open an Issue with the MCAP Project: The best course of action now is to report this issue to the developers of mcap. Provide them with the following details:

Your macOS version and architecture (e.g., macOS 14 Sonoma, arm64).
The output of mcap --version.
The error message you're getting.
The fact that the file header appears valid (od output).
The size of your output.mcap file.
They can investigate further and might be able to provide a solution or workaround.

Try a Different Environment: If you have access to another computer (or a virtual machine), try installing mcap there and see if it can open the file. This can help rule out any system-specific issues.

```

james-rms, 2024-07-17:
hi @josephykwang - the [spec](https://mcap.dev/spec) dictates that there should be a [header record](https://mcap.dev/spec#header-op0x01) after the initial 8-byte MCAP magic. Your write code writes the magic, then prints "written MCAP header.", but no header record is present.

There are several other issues with your writer code - there are no record opcodes being written, the record lengths are 8-byte integers, not 4-bytes, and there need to be schema and channel records for any messages you write.

Consider using the python writer API to handle this for you: https://mcap.dev/docs/python/ with `pip install mcap`

closed 2024-07-17 by james-rms

josephykwang, 2024-07-17:
change the code to ```import os
import shutil

import json

import subprocess

from mcap.writer import Writer

# Define the MCAP file paths

frames_dir = './output_frames'

output_h264_file = './output.h264'

mcap_file = './mcap_files/output.mcap'

# Function to convert PNG frames to H.264

def convert_png_to_h264(frames_dir, output_h264_file, frame_rate):

    frame_files = sorted([os.path.join(frames_dir, f) for f in os.listdir(frames_dir) if f.endswith('.png')])

    

    if not frame_files:

        print("No PNG frames found in the directory.")

        return

    # Use ffmpeg to convert PNG frames to H.264

    input_files = os.path.join(frames_dir, 'frame_%04d.png')

    subprocess.run(['ffmpeg', '-framerate', str(frame_rate), '-i', input_files, '-c:v', 'libx264', '-pix_fmt', 'yuv420p', output_h264_file], check=True)

# Function to process H.264 file into custom MCAP format

def process_h264_to_mcap_custom(h264_file, mcap_file, width, height, frame_rate):

    json_schema = {

        "type": "object",

        "properties": {

            "timestamp": {

                "type": "object",

                "properties": {

                    "sec": {"type": "integer"},

                    "nsec": {"type": "integer"},

                },

                "required": ["sec", "nsec"],

            },

            "frame_id": {"type": "string"},

            "data": {"type": "string"},  # Base64 encoded H.264 frame data

        },

        "required": ["timestamp", "frame_id", "data"],

    }

    with open(h264_file, 'rb') as h264, open(mcap_file, 'wb') as f:

        writer = Writer(f)

        

        # Write the MCAP header, schema, and channel

        writer.start(profile="", library="python-mcap")

        schema_id = writer.register_schema(

            name="foxglove.CompressedImage",

            encoding="jsonschema",

            data=json.dumps(json_schema).encode("utf-8"),

        )

        channel_id = writer.register_channel(

            topic="/image/compressed",

            message_encoding="json",

            schema_id=schema_id,

            metadata={

                "format": "h264",  

                "width": str(width),

                "height": str(height),

                "frame_rate": str(frame_rate),

            },

        )

        

        # Write the frame data

        while True:

            frame_chunk = h264.read(512*1024)  # 512 KB chunks

            if not frame_chunk:

                break

            message = {

                "channel_id": channel_id,

                "log_time": 0,

                "publish_time": 0,

                "data": frame_chunk

            }

            writer.add_message(**message)

        

        writer.finish()

        print(f"Finished writing MCAP file: {mcap_file}")

# Ensure the frames are in the correct directory

if not os.path.exists(frames_dir):

    os.makedirs(frames_dir)

# Convert the frames to MCAP files

convert_png_to_h264(frames_dir, output_h264_file, frame_rate=30)

process_h264_to_mcap_custom(output_h264_file, mcap_file, width=1936, height=1220, frame_rate=30)
``` 

% mcap doctor output.mcap 

% mcap info output.mcap

library:   python-mcap  

profile:                

messages:  1            

duration:  0s           

start:     0.000000000  

end:       0.000000000  

compression:

 zstd: [1/1 chunks] [382.40 KiB/382.42 KiB (-0.00%)] 

channels:

 (1) /image/compressed  1 msgs   : foxglove.CompressedImage [jsonschema]  

attachments: 0

metadata: 0

when loading output.mcap onto foxglove studio, see ===Check that your input file is not corrupted.=== 
any suggestioin?    On Wednesday, July 17, 2024 at 04:15:56 PM PDT, james-rms ***@***.***> wrote:  
 
 

Closed #1198 as completed.

—
Reply to this email directly, view it on GitHub, or unsubscribe.
You are receiving this because you were mentioned.Message ID: ***@***.***>

james-rms, 2024-07-17:
Your code uses a JSON schema, but the compressed image messages are not JSON-encoded - they appear to be raw h.264 bytes. You need to ensure that your messages are encoded according to the schema you declare in the file. 

Also, if you're using h.264, you need to use CompressedVideo messages rather than CompressedImage:

https://docs.foxglove.dev/docs/visualization/message-schemas/compressed-video

Note that per [that schema](https://github.com/foxglove/schemas/blob/main/schemas/jsonschema/CompressedVideo.json), the h.264 data needs to be base64-encoded inside the message object.

josephykwang, 2024-07-18:
thanks for the recommendation and prompt response.

changed the code ```
import os

import shutil

import json

import subprocess

from mcap.writer import Writer

import base64

from datetime import datetime, timezone

# Define the MCAP file paths

frames_dir = './output_frames'

output_h264_file = './output.h264'

mcap_file = './output.mcap'

# Ensure the frames are in the correct directory

if not os.path.exists(frames_dir):

    os.makedirs(frames_dir)

# Function to convert PNG frames to H.264

def convert_png_to_h264(frames_dir, output_h264_file, frame_rate):

    frame_files = sorted([os.path.join(frames_dir, f) for f in os.listdir(frames_dir) if f.endswith('.png')])

    

    if not frame_files:

        print("No PNG frames found in the directory.")

        return

    # Use ffmpeg to convert PNG frames to H.264

    input_files = os.path.join(frames_dir, 'frame_%04d.png')

    subprocess.run(['ffmpeg', '-framerate', str(frame_rate), '-i', input_files, '-c:v', 'libx264', '-pix_fmt', 'yuv420p', output_h264_file], check=True)

# Function to process H.264 file into MCAP format

def process_h264_to_mcap(h264_file, mcap_file, width, height, frame_rate):

    with open(h264_file, 'rb') as h264, open(mcap_file, 'wb') as f:

        writer = Writer(f)

        # Initialize MCAP header and schemas

        writer.start(profile="", library="python-mcap")

        schema_id = writer.register_schema(

            name="foxglove.CompressedVideo",

            encoding="jsonschema",

            data=json.dumps({
[... 231 lines cut ...]
channels:

 (1) /video/compressed  8 msgs (34.29 Hz)   : foxglove.CompressedVideo [jsonschema]  

attachments: 0

metadata: 0
=== 
don't see image

    On Wednesday, July 17, 2024 at 04:57:20 PM PDT, james-rms ***@***.***> wrote:  
 
 

Your code uses a JSON schema, but the compressed image messages are not JSON-encoded - they appear to be raw h.264 bytes. You need to ensure that your messages are encoded according to the schema you declare in the file.

Also, if you're using h.264, you need to use CompressedVideo messages rather than CompressedImage:

https://docs.foxglove.dev/docs/visualization/message-schemas/compressed-video

Note that per that schema, the h.264 data needs to be base64-encoded inside the message object.

—
Reply to this email directly, view it on GitHub, or unsubscribe.
You are receiving this because you were mentioned.Message ID: ***@***.***>

james-rms, 2024-07-18:
@josephykwang i'm not sure if i'm reading your ffmpeg output right, but your video stream appears to include B-frames, which are not supported: https://docs.foxglove.dev/docs/visualization/message-schemas/compressed-video#data

josephykwang, 2024-07-18:
changed the code so that ffmpeg command is modified to include the -x264opts keyint=1 option, which ensures that only I-frames and P-frames are used, and B-frames are excluded.```
python3 ./frames_to_mcap40.py 

ffmpeg version 4.1.11-0+deb10u1 Copyright (c) 2000-2023 the FFmpeg developers

  built with gcc 8 (Debian 8.3.0-6)

  configuration: --prefix=/usr --extra-version=0+deb10u1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared

  libavutil      56. 22.100 / 56. 22.100

  libavcodec     58. 35.100 / 58. 35.100

  libavformat    58. 20.100 / 58. 20.100

  libavdevice    58.  5.100 / 58.  5.100

  libavfilter     7. 40.101 /  7. 40.101

  libavresample   4.  0.  0 /  4.  0.  0

  libswscale      5.  3.100 /  5.  3.100

  libswresample   3.  3.100 /  3.  3.100

  libpostproc    55.  3.100 / 55.  3.100

Input #0, image2, from './output_frames/frame_%04d.png':

  Duration: 00:00:03.30, start: 0.000000, bitrate: N/A

    Stream #0:0: Video: png, rgb24(pc), 1936x1220, 30 fps, 30 tbr, 30 tbn, 30 tbc

Stream mapping:

  Stream #0:0 -> #0:0 (png (native) -> h264 (libx264))

Press [q] to stop, [?] for help

[libx264 @ 0x557a3c6a7240] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2 AVX512

[libx264 @ 0x557a3c6a7240] profile High, level 5.0

Output #0, h264, to './output.h264':

  Metadata:

    encoder         : Lavf58.20.100

    Stream #0:0: Video: h264 (libx264), yuv420p, 1936x1220, q=-1--1, 30 fps, 30 tbn, 30 tbc

    Metadata:

      encoder         : Lavc58.35.100 libx264

    Side data:

      cpb: bitrate max/min/avg: 0/0/0 buffer size: 0 vbv_delay: -1

frame=   99 fps= 72 q=-1.0 Lsize=    6626kB time=00:00:03.30 bitrate=16449.0kbits/s speed= 2.4x    

video:6626kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%

[libx264 @ 0x557a3c6a7240] frame I:99    Avg QP:26.24  size: 68537

[libx264 @ 0x557a3c6a7240] mb I  I16..4: 15.6% 71.6% 12.8%

[libx264 @ 0x557a3c6a7240] 8x8 transform intra:71.6%

[libx264 @ 0x557a3c6a7240] coded y,uvDC,uvAC intra: 61.5% 25.3% 0.8%
[... 234 lines cut ...]

  libswresample   5.  1.100 /  5.  1.100

  libpostproc    58.  1.100 / 58.  1.100

Input #0, h264, from 'output.h264':

  Duration: N/A, bitrate: N/A

  Stream #0:0: Video: h264 (High), yuv420p(progressive), 1936x1220, 25 fps, 60 tbr, 1200k tbn

[swscaler @ 0x140e70000] No accelerated colorspace conversion found from yuv420p to bgr24.

[swscaler @ 0x141558000] No accelerated colorspace conversion found from yuv420p to bgr24.

[swscaler @ 0x141c40000] No accelerated colorspace conversion found from yuv420p to bgr24.

[swscaler @ 0x142328000] No accelerated colorspace conversion found from yuv420p to bgr24.

[swscaler @ 0x142a10000] No accelerated colorspace conversion found from yuv420p to bgr24.

[swscaler @ 0x1430f8000] No accelerated colorspace conversion found from yuv420p to bgr24.

[swscaler @ 0x1437e0000] No accelerated colorspace conversion found from yuv420p to bgr24.
===

josephykwang, 2024-07-18:
see the following on foxglove studio. any idea how to fix it?

james-rms, 2024-07-19:
Your comment is hard to parse, but as I understand it you don't see an error in visualization, but you also don't see an image.  if you provide the resulting MCAP file it's possible @snosenzo could take a look and figure out why.

josephykwang, 2024-07-19:
the mail bounced back stating the file is too big for github.

josephykwang, 2024-07-19:
here is the python code
```
import os

import shutil

import json

import subprocess

from mcap.writer import Writer

import base64

from datetime import datetime, timezone

# Define the MCAP file paths

frames_dir = './output_frames'

output_h264_file = './output.h264'

mcap_file = './output.mcap'

# Ensure the frames are in the correct directory

if not os.path.exists(frames_dir):

    os.makedirs(frames_dir)

# Function to convert PNG frames to H.264 without B-frames

def convert_png_to_h264(frames_dir, output_h264_file, frame_rate):

    frame_files = sorted([os.path.join(frames_dir, f) for f in os.listdir(frames_dir) if f.endswith('.png')])

    

    if not frame_files:

        print("No PNG frames found in the directory.")

        return

    # Use ffmpeg to convert PNG frames to H.264 without B-frames

    input_files = os.path.join(frames_dir, 'frame_%04d.png')

    subprocess.run(['ffmpeg', '-framerate', str(frame_rate), '-i', input_files, '-c:v', 'libx264', '-pix_fmt', 'yuv420p', '-x264opts', 'keyint=1', output_h264_file], check=True)

# Function to process H.264 file into MCAP format

def process_h264_to_mcap(h264_file, mcap_file, width, height, frame_rate):

    with open(h264_file, 'rb') as h264, open(mcap_file, 'wb') as f:

        writer = Writer(f)

        # Initialize MCAP header and schemas

        writer.start(profile="", library="python-mcap")

        schema_id = writer.register_schema(

            name="foxglove.CompressedVideo",

            encoding="jsonschema",

            data=json.dumps({

                "type": "object",

                "properties": {

                    "timestamp": {

                        "type": "object",

                        "properties": {

                            "sec": {"type": "integer"},

                            "nsec": {"type": "integer"}

                        },

                        "required": ["sec", "nsec"]

                    },

                    "frame_id": {"type": "string"},

                    "data": {"type": "string"},  # Base64 encoded H.264 frame data

                },

                "required": ["timestamp", "frame_id", "data"]

            }).encode("utf-8"),

        )

        channel_id = writer.register_channel(

            topic="/video/compressed",

            message_encoding="json",

            schema_id=schema_id,

            metadata={

                "format": "h264",

                "width": str(width),

                "height": str(height),

                "frame_rate": str(frame_rate),

            }

        )

        # Process and write frame data with timestamps

        frame_id = 0

        now = datetime.now(timezone.utc)  # Reference timestamp

        while True:

            frame_data = h264.read(512 * 1024) 

            if not frame_data:

                break

            # Calculate frame timestamp based on frame rate

            sec = int(frame_id / frame_rate)

            nsec = int((frame_id % frame_rate) * (10**9 / frame_rate))

            message = {

                "channel_id": channel_id,

                "log_time": int(now.timestamp() * 10**9 + sec * 10**9 + nsec),

                "publish_time": 0,

                "data": json.dumps({

                    "timestamp": {"sec": sec, "nsec": nsec},

                    "frame_id": str(frame_id),

                    "data": base64.b64encode(frame_data).decode('utf-8')

                }).encode('utf-8')  # Convert to bytes

            }

            writer.add_message(**message)

            frame_id += 1

        writer.finish()

        print(f"Finished writing MCAP file: {mcap_file}")

# Convert the frames to MCAP files

convert_png_to_h264(frames_dir, output_h264_file, frame_rate=30)
process_h264_to_mcap(output_h264_file, mcap_file, width=1936, height=1220, frame_rate=30)```    On Thursday, July 18, 2024 at 07:39:37 PM PDT, Joseph Wang ***@***.***> wrote:  
 
  the mail bounced back stating the file is too big for github.
    On Thursday, July 18, 2024 at 06:15:33 PM PDT, james-rms ***@***.***> wrote:  
 
 

Your comment is hard to parse, but as I understand it you don't see an error in visualization, but you also don't see an image. if you provide the resulting MCAP file it's possible @snosenzo could take a look and figure out why.

—
Reply to this email directly, view it on GitHub, or unsubscribe.
You are receiving this because you were mentioned.Message ID: ***@***.***>

josephykwang, 2024-07-19:
uploaded files, code, and output_frames/*.png to https://drive.google.com/drive/u/1/folders/163E1sNpfXuPuHjnOtULIt-ubE6DooNRh 
***@***.******@***.******@***.*** access
    On Thursday, July 18, 2024 at 07:45:53 PM PDT, Joseph Wang ***@***.***> wrote:  
 
  here is the python code
```
import os

import shutil

import json

import subprocess

from mcap.writer import Writer

import base64

from datetime import datetime, timezone

# Define the MCAP file paths

frames_dir = './output_frames'

output_h264_file = './output.h264'

mcap_file = './output.mcap'

# Ensure the frames are in the correct directory

if not os.path.exists(frames_dir):

    os.makedirs(frames_dir)

# Function to convert PNG frames to H.264 without B-frames

def convert_png_to_h264(frames_dir, output_h264_file, frame_rate):

    frame_files = sorted([os.path.join(frames_dir, f) for f in os.listdir(frames_dir) if f.endswith('.png')])

    

    if not frame_files:

        print("No PNG frames found in the directory.")

        return

    # Use ffmpeg to convert PNG frames to H.264 without B-frames

    input_files = os.path.join(frames_dir, 'frame_%04d.png')

    subprocess.run(['ffmpeg', '-framerate', str(frame_rate), '-i', input_files, '-c:v', 'libx264', '-pix_fmt', 'yuv420p', '-x264opts', 'keyint=1', output_h264_file], check=True)

# Function to process H.264 file into MCAP format

def process_h264_to_mcap(h264_file, mcap_file, width, height, frame_rate):

    with open(h264_file, 'rb') as h264, open(mcap_file, 'wb') as f:

        writer = Writer(f)

        # Initialize MCAP header and schemas

        writer.start(profile="", library="python-mcap")

        schema_id = writer.register_schema(

            name="foxglove.CompressedVideo",

            encoding="jsonschema",

            data=json.dumps({

                "type": "object",

                "properties": {

                    "timestamp": {

                        "type": "object",

                        "properties": {

                            "sec": {"type": "integer"},

                            "nsec": {"type": "integer"}

                        },

                        "required": ["sec", "nsec"]

                    },

                    "frame_id": {"type": "string"},

                    "data": {"type": "string"},  # Base64 encoded H.264 frame data

                },

                "required": ["timestamp", "frame_id", "data"]

            }).encode("utf-8"),

        )

        channel_id = writer.register_channel(

            topic="/video/compressed",

            message_encoding="json",

            schema_id=schema_id,

            metadata={

                "format": "h264",

                "width": str(width),

                "height": str(height),

                "frame_rate": str(frame_rate),

            }

        )

        # Process and write frame data with timestamps

        frame_id = 0

        now = datetime.now(timezone.utc)  # Reference timestamp

        while True:

            frame_data = h264.read(512 * 1024) 

            if not frame_data:

                break

            # Calculate frame timestamp based on frame rate

            sec = int(frame_id / frame_rate)

            nsec = int((frame_id % frame_rate) * (10**9 / frame_rate))

            message = {

                "channel_id": channel_id,

                "log_time": int(now.timestamp() * 10**9 + sec * 10**9 + nsec),

                "publish_time": 0,

                "data": json.dumps({

                    "timestamp": {"sec": sec, "nsec": nsec},

                    "frame_id": str(frame_id),

                    "data": base64.b64encode(frame_data).decode('utf-8')

                }).encode('utf-8')  # Convert to bytes

            }

            writer.add_message(**message)

            frame_id += 1

        writer.finish()

        print(f"Finished writing MCAP file: {mcap_file}")

# Convert the frames to MCAP files

convert_png_to_h264(frames_dir, output_h264_file, frame_rate=30)
process_h264_to_mcap(output_h264_file, mcap_file, width=1936, height=1220, frame_rate=30)```    On Thursday, July 18, 2024 at 07:39:37 PM PDT, Joseph Wang ***@***.***> wrote:  
 
  the mail bounced back stating the file is too big for github.
    On Thursday, July 18, 2024 at 06:15:33 PM PDT, james-rms ***@***.***> wrote:  
 
 

Your comment is hard to parse, but as I understand it you don't see an error in visualization, but you also don't see an image. if you provide the resulting MCAP file it's possible @snosenzo could take a look and figure out why.

—
Reply to this email directly, view it on GitHub, or unsubscribe.
You are receiving this because you were mentioned.Message ID: ***@***.***>

josephykwang, 2024-07-22:
any suggestion?
    On Friday, July 19, 2024 at 09:04:55 AM PDT, Joseph Wang ***@***.***> wrote:  
 
  uploaded files, code, and output_frames/*.png to https://drive.google.com/drive/u/1/folders/163E1sNpfXuPuHjnOtULIt-ubE6DooNRh 
***@***.******@***.******@***.*** access
    On Thursday, July 18, 2024 at 07:45:53 PM PDT, Joseph Wang ***@***.***> wrote:  
 
  here is the python code
```
import os

import shutil

import json

import subprocess

from mcap.writer import Writer

import base64

from datetime import datetime, timezone

# Define the MCAP file paths

frames_dir = './output_frames'

output_h264_file = './output.h264'

mcap_file = './output.mcap'

# Ensure the frames are in the correct directory

if not os.path.exists(frames_dir):

    os.makedirs(frames_dir)

# Function to convert PNG frames to H.264 without B-frames

def convert_png_to_h264(frames_dir, output_h264_file, frame_rate):

    frame_files = sorted([os.path.join(frames_dir, f) for f in os.listdir(frames_dir) if f.endswith('.png')])

    

    if not frame_files:

        print("No PNG frames found in the directory.")

        return

    # Use ffmpeg to convert PNG frames to H.264 without B-frames

    input_files = os.path.join(frames_dir, 'frame_%04d.png')

    subprocess.run(['ffmpeg', '-framerate', str(frame_rate), '-i', input_files, '-c:v', 'libx264', '-pix_fmt', 'yuv420p', '-x264opts', 'keyint=1', output_h264_file], check=True)

# Function to process H.264 file into MCAP format

def process_h264_to_mcap(h264_file, mcap_file, width, height, frame_rate):

    with open(h264_file, 'rb') as h264, open(mcap_file, 'wb') as f:

        writer = Writer(f)

        # Initialize MCAP header and schemas

        writer.start(profile="", library="python-mcap")

        schema_id = writer.register_schema(

            name="foxglove.CompressedVideo",

            encoding="jsonschema",

            data=json.dumps({

                "type": "object",

                "properties": {

                    "timestamp": {

                        "type": "object",

                        "properties": {

                            "sec": {"type": "integer"},

                            "nsec": {"type": "integer"}

                        },

                        "required": ["sec", "nsec"]

                    },

                    "frame_id": {"type": "string"},

                    "data": {"type": "string"},  # Base64 encoded H.264 frame data

                },

                "required": ["timestamp", "frame_id", "data"]

            }).encode("utf-8"),

        )

        channel_id = writer.register_channel(

            topic="/video/compressed",

            message_encoding="json",

            schema_id=schema_id,

            metadata={

                "format": "h264",

                "width": str(width),

                "height": str(height),

                "frame_rate": str(frame_rate),

            }

        )

        # Process and write frame data with timestamps

        frame_id = 0

        now = datetime.now(timezone.utc)  # Reference timestamp

        while True:

            frame_data = h264.read(512 * 1024) 

            if not frame_data:

                break

            # Calculate frame timestamp based on frame rate

            sec = int(frame_id / frame_rate)

            nsec = int((frame_id % frame_rate) * (10**9 / frame_rate))

            message = {

                "channel_id": channel_id,

                "log_time": int(now.timestamp() * 10**9 + sec * 10**9 + nsec),

                "publish_time": 0,

                "data": json.dumps({

                    "timestamp": {"sec": sec, "nsec": nsec},

                    "frame_id": str(frame_id),

                    "data": base64.b64encode(frame_data).decode('utf-8')

                }).encode('utf-8')  # Convert to bytes

            }

            writer.add_message(**message)

            frame_id += 1

        writer.finish()

        print(f"Finished writing MCAP file: {mcap_file}")

# Convert the frames to MCAP files

convert_png_to_h264(frames_dir, output_h264_file, frame_rate=30)
process_h264_to_mcap(output_h264_file, mcap_file, width=1936, height=1220, frame_rate=30)```    On Thursday, July 18, 2024 at 07:39:37 PM PDT, Joseph Wang ***@***.***> wrote:  
 
  the mail bounced back stating the file is too big for github.
    On Thursday, July 18, 2024 at 06:15:33 PM PDT, james-rms ***@***.***> wrote:  
 
 

Your comment is hard to parse, but as I understand it you don't see an error in visualization, but you also don't see an image. if you provide the resulting MCAP file it's possible @snosenzo could take a look and figure out why.

—
Reply to this email directly, view it on GitHub, or unsubscribe.
You are receiving this because you were mentioned.Message ID: ***@***.***>

james-rms, 2024-07-22:
Hi joseph, I'm afraid I don't know how to help you further in this context. If you ask in our community discord https://discord.com/channels/1240935776649478144/1245114173776597024 (join via https://foxglove.dev/chat first) someone may be able to help you.