foxglove/mcap #1224: Python MCAP Reader doesn't work with 3.8 and before.

JMarple, 2024-08-23:
**Description**
When I try to import from mcap_protobuf.reader in python (as per the mcap read example here: https://github.com/foxglove/mcap/blob/main/python/examples/raw/reader.py) I get the following error: 

```
File "/home/jmarple/.local/lib/python3.8/site-packages/mcap_protobuf/reader.py", line 27, in <module>
    source: Union[str, bytes, PathLike[str], McapReader, IO[bytes]],
TypeError: 'ABCMeta' object is not subscriptable
```

- Version: pip3 show mcap shows V1.1.0 & Python 3.8.10
- Platform: Ubuntu 20.04 

**Steps To Reproduce**
Open up a Python 3.8 terminal and run `from mcap_protobuf.reader import make_reader`

JMarple, 2024-08-23:
Trying a few different versions, it seems you can run the example fine in 3.9.9, 3.10.9 and 3.11.9.  It appears 3.8.10 and before this breaks.  

I figure the solution is either fix the typing error, or update the minimum python version on pip to 3.9

james-rms, 2024-08-26:
This is surprising - we build and run the examples in CI using python 3.8, so I would not expect those to work.

I also tried this locally:
```
$ cat Dockerfile 
FROM python:3.8

COPY . /mcap

WORKDIR /mcap

RUN pip install -e mcap
RUN pip install -e mcap-protobuf-support
$ docker run --rm -it $(docker build -q .) python3
Python 3.8.19 (default, Aug 13 2024, 19:21:02) 
[GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from mcap_protobuf.reader import make_reader
>>> make_reader
<function make_reader at 0xffffb67c58b0>
>>> 
```

JMarple, 2024-08-26:
Hmm, that's interesting.  I was able to reproduce with a fresh python install using `pyenv`. 

`pyenv install 3.8.19`
`pyenv local 3.8.19` 
`python3 --version` shows 3.8.19
`pip3 install mcap mcap-protobuf-support`

The error reproduces the same as above.

james-rms, 2024-08-27:
I had some trouble getting pyenv up and running and also this is not a pyenv-specific issue (since you reported this initially without pyenv) so I don't feel like it'd be productive to pursue that. I feel pretty confident that the issue you're seeing has to do with your python installation or environment, because this code works fine using the official python 3.8 docker container and also in CI.

closed 2024-08-27 by james-rms