foxglove/mcap #1268: `mcap filter` with `-e`, `-s` not behaving correctly

mwcondino, 2024-11-12:
**Description**

- Version: `v0.0.49`
- Platform: Ubuntu 22 (in a docker container).

**Steps To Reproduce**
- Download the attached test bag `test.mcap` into local directory
- Run the following command:
```
mcap filter ./test.mcap -o ./test_out.mcap --start-secs 1727916694 --end-secs 1727916724
```

This fails with `configuration error: invalid time range query, end-time is before start-time` and exit code `1`.

**Expected Behavior**
No issue with this filter, as the given times are valid.

**Workaround**
It appears that using `-E` and `-S` works, but requires converting the time into nanoseconds. This is fine for my application. However, I was surprised that `-s` and `-e` stopped working; on my previous version (v0.0.42), there were no issues with these args.

bennetthardwick, 2024-11-12:
Looks like it might have been a bad merge. I don't think this line should be there anymore: https://github.com/foxglove/mcap/blob/28627b721622183e59f7be8032cbf33033558a23/go/cli/mcap/cmd/filter.go#L80-L82

mwcondino, 2024-11-12:
> Looks like it might have been a bad merge. I don't think this line should be there anymore:
> 
> https://github.com/foxglove/mcap/blob/28627b721622183e59f7be8032cbf33033558a23/go/cli/mcap/cmd/filter.go#L80-L82

Yeah I'm not familiar with the changes from the last release, but this blurb from the release notes stuck out to me:

> filter: Added --start and --end arguments that accept either nanosecond or RFC3339-formatted log times.

I just tested on `v0.0.48`; no issue with that release.

closed 2024-11-13 by james-rms

stewpend0us, 2024-11-13:
I'm having this issue with v0.0.49 also.

james-rms, 2024-11-13:
@stewpend0us v0.0.50 fixes this issue.

mwcondino, 2024-11-13:
Thanks for the fix @james-rms!

PR #1269 (2024-11-13): filter: fix end second parsing
### Changelog
- Fixed: the `-e` argument to `mcap filter` will be interpreted properly again.

### Docs

None.
### Description

The last change to `filter` mistakenly swapped the `nanos` and `seconds` arguments to `parseTimestampArgs` for the filter end time.

### Testing
- Tested `mcap filter -s 10 -e 1000` manually, previously failed, now works.

Fixes: #1268
commit 77ad4b2d (2024-11-13): filter: fix end second parsing (#1269)

### Changelog
- Fixed: the `-e` argument to `mcap filter` will be interpreted properly
again.

### Docs

None.
### Description

The last change to `filter` mistakenly swapped the `nanos` and `seconds`
arguments to `parseTimestampArgs` for the filter end time.

### Testing
- Tested `mcap filter -s 10 -e 1000` manually, previously failed, now
works.

Fixes: #1268