foxglove/mcap #1271: Failing to clone foxglove mcap due to lack of lfs storage

Ryang20718, 2024-11-17:
```
git clone https://github.com/foxglove/mcap.git
Cloning into 'mcap'...
remote: Enumerating objects: 125415, done.
remote: Counting objects: 100% (1073/1073), done.
remote: Compressing objects: 100% (533/533), done.
remote: Total 125415 (delta 635), reused 878 (delta 510), pack-reused 124342 (from 1)
Receiving objects: 100% (125415/125415), 133.15 MiB | 34.04 MiB/s, done.
Resolving deltas: 100% (106570/106570), done.
Downloading .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs (1.1 MB)
Error downloading object: .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs (72805e7): Smudge error: Error downloading .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs (72805e7a83dd57763f1707ab4fa6dbd5391a60f8271fc33fb55a6cd2a55b1143): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

Errors logged to '/tmp/mcap/.git/lfs/logs/20241117T203730.628917553.log'.
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'

```

amacneil, 2024-11-17:
Whoops, looks like our payment failed. Also I thought open source repos had free LFS storage - I guess not.

Try again now.

closed 2024-11-17 by amacneil

jtbandes, 2024-11-17:
> Every account using Git Large File Storage receives 1 GiB of free storage and 1 GiB a month of free bandwidth.