foxglove/mcap #1276: end_time excludes messages that occur at the same time

airalcorn2, 2024-11-20:
**Description**

- Version: 1.2.1
- Platform: Ubuntu 22.04

**Steps To Reproduce**

```python
from mcap.reader import make_reader
from mcap_ros2.decoder import DecoderFactory

mcap_path = "some_mcap.mcap"
some_topic = "some_topic"
first_ns = None
with open(mcap_path, "rb") as mcap_f:
    reader = make_reader(mcap_f, decoder_factories=[DecoderFactory()])
    for _, _, ros_msg in reader.iter_messages(topics=[some_topic]):
        first_ns = ros_msg.log_time
        break

first_msg = None
with open(mcap_path, "rb") as mcap_f:
    reader = make_reader(mcap_f, decoder_factories=[DecoderFactory()])
    for _, channel, _, ros_msg in reader.iter_decoded_messages(
        topics=[some_topic], start_time=first_ns, end_time=first_ns
    ):
        first_msg = ros_msg
        break

# Error because first_msg is None.
print(first_msg.header)

with open(mcap_path, "rb") as mcap_f:
    reader = make_reader(mcap_f, decoder_factories=[DecoderFactory()])
    for _, channel, _, ros_msg in reader.iter_decoded_messages(
        topics=[some_topic], start_time=first_ns, end_time=first_ns + 1
    ):
        first_msg = ros_msg
        break

# Correct message.
print(first_msg.header)
```

**Expected Behavior**

[image]

According to the documentation [here](https://mcap.dev/docs/python/mcap-apidoc/mcap.reader#mcap.reader.McapReader.iter_decoded_messages), messages that were logged either _before_ `start_time` or _after_ `end_time` aren't returned by the iterator, but the iterator appears to also exclude messages that were logged _at the same time_ as `end_time`.

jtbandes, 2024-11-20:
Hmm, could it be as simple as changing these `>=` to `>`?

https://github.com/foxglove/mcap/blob/c8afa1eaec756956118cee6677981ee424d2eee0/python/mcap/mcap/reader.py#L314-L315

https://github.com/foxglove/mcap/blob/c8afa1eaec756956118cee6677981ee424d2eee0/python/mcap/mcap/reader.py#L93-L94

exogen, 2024-11-20:
> And I guess the question is whether the API was really intended to be inclusive or exclusive...

Change looks simple but yeah, the unfortunate part of making it inclusive (even though it matches the docs) is that you can't use an `end_time` as the next `start_time` if you want to iterate in time slices, since that'd duplicate records, so you'd need to do the +1ns trick. If it were up to me, I'd keep end time exclusive and update the docs, especially since it wouldn't be a breaking change to the library.

rdumas01, 2024-12-02:
Purely based on Python's usual behavior with bounds, I would also lean toward simply updating the docs (`list[start:end]`, `array[start:end]`, `range(start, end)`, `numpy.arange(start, end)`, etc... are all exclusive)

closed 2024-12-11 by linear[bot]