foxglove/mcap #1326: C++: Linker errors when building as a static library on Windows

asahtik, 2025-02-04:
**Description**

- Version: commit 0828dedf962416d472fbcd8e0ae0ad3c86e7cae9
- Platform: Windows 10, C++

I am attempting to build mcap as a static library and link it to a project (similarly to [mcap_builder](https://github.com/olympus-robotics/mcap_builder) where I ran into the same issue). Everything works as expected on Linux, however on Windows I get multiple warnings like the following:
```
C:\Users\user\Documents\mcap_win_err_mre\mcap\include\mcap\writer.hpp(475,32): warning C4251: 'mcap
::McapWriter::attachmentIndex_': class 'std::vector<mcap::AttachmentIndex,std::allocator<mcap::AttachmentInde
x>>' needs to have dll-interface to be used by clients of class 'mcap::McapWriter' [C:\Users\user\Documents\mcap_win_err_mre\build\mcap.vcxproj]
```

followed by linker errors:
```
mcap_test.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: void __cdecl mcap::M
capReader::close(void)" (__imp_?close@McapReader@mcap@@QEAAXXZ) referenced in function main [C:\Users\user\
Documents\mcap_win_err_mre\build\mcap_test.vcxproj]
```

I only get errors if I build statically. If I change `add_library (mcap STATIC "mcap.cpp")` to `add_library (mcap SHARED "mcap.cpp")` everything seems to work as expected.

[Possibly related issue](https://github.com/olympus-robotics/mcap_builder/issues/8)

**Steps To Reproduce**

[MRE](https://github.com/asahtik/mcap_win_err_mre/tree/master)

On Windows:
```
git clone https://github.com/asahtik/mcap_win_err_mre.git
cd mcap_win_err_mre
cmake -S. -Bbuild
cmake --build build
```

**Expected Behavior**

Code compiles and links successfully without errors

james-rms, 2025-02-28:
You're running into a known issue of trying to build a static library which exports STL-templated classes. The MCAP library doesn't support this at the moment, but it could be added by adding explicit template instantiations of all of these return types, as is explained here: https://jeffpar.github.io/kbarchive/kb/168/Q168958/

I'd be happy to review a PR with those changes - I don't have a windows machine to experiment with this on right now.

SXongin, 2025-10-08:
I think this issue can be fixed by pr #1431  (not released yet)
if you are wrapping mcap in a  static library in windows, just define MCAP_PUBLIC as empty before `include` mcap headers
```cpp
#define MCAP_PUBLIC
#include <mcap/writer.hpp>
#include <mcap/reader.hpp>
```