foxglove/mcap #1337: `mcap_reader.read_ros2_messages` fails when reading a message containing an array of int8's

kwesiRutledge-aescape, 2025-02-24:
**Description**

While defining some analysis scripts for our `.mcap` data, we received an unexpected error while trying to read the bag files. It appears that when the `mcap_reader.read_ros2_messages` function encounters a field containing 7 int8's in an `int8[]` variable, that it causes an error to be raised and our scripts to fail.

The error is:
```
Traceback (most recent call last):
  File "/workspace/install/test_runner/lib/python3.10/site-packages/XXXX/YYYY.py", line 162, in read_bag_dir
    for msg in mcap_reader.read_ros2_messages(bag_file, topics):
  File "/usr/local/lib/python3.10/dist-packages/mcap_ros2/reader.py", line 74, in read_ros2_messages
    for schema, channel, message, ros2_msg in reader.iter_decoded_messages(
  File "/usr/local/lib/python3.10/dist-packages/mcap/reader.py", line 193, in iter_decoded_messages
    schema, channel, message, decoded_message(schema, channel, message)
  File "/usr/local/lib/python3.10/dist-packages/mcap/reader.py", line 184, in decoded_message
    return decoder(message.data)
  File "/usr/local/lib/python3.10/dist-packages/mcap_ros2/_dynamic.py", line 240, in <lambda>
    return lambda data: read_message(schema_name, msgdefs, data)
  File "/usr/local/lib/python3.10/dist-packages/mcap_ros2/_dynamic.py", line 213, in read_message
    return _read_complex_type(msgdef, msgdefs, reader)
  File "/usr/local/lib/python3.10/dist-packages/mcap_ros2/_dynamic.py", line 323, in _read_complex_type
    else reader.sequence_length()
  File "/usr/local/lib/python3.10/dist-packages/mcap_ros2/_cdr.py", line 179, in sequence_length
    return self.uint32()
  File "/usr/local/lib/python3.10/dist-packages/mcap_ros2/_cdr.py", line 119, in uint32
    return self._unpack(
  File "/usr/local/lib/python3.10/dist-packages/mcap_ros2/_cdr.py", line 256, in _unpack
    value = fn(self.data, self.offset)[0]
struct.error: unpack_from requires a buffer of at least 56 bytes for unpacking 4 bytes at offset 52 (actual buffer size is 52)
```

- Version: `mcap` version 1.2.2 and `mcap-ros2-support` version 0.5.2
- Platform: Linux

**Steps To Reproduce**

1. Define a custom message type that contains int8[]'s within it. For example:
```
std_msgs/Header header

int8 solver_status
float64 solver_obj

int8[] joint_health_flag
```
2. Save several of these messages to an `mcap` file.
3. Attempt to read the messages using `mcap_ros2.mcap_reader.read_ros2_messages`.

**Expected Behavior**

We expected this method to successfully read all messages from the `.mcap` file, but instead we receive the error above.

jtbandes, 2025-02-24:
Hi there,
Could you provide better steps to reproduce? When I followed your instructions, I was able to successfully read the generated mcap file:

```py
import rclpy
from test_pkg.msg import McapGh1337
from rclpy.serialization import serialize_message
import rosbag2_py
from array import array
import sys
from mcap_ros2.decoder import DecoderFactory
from mcap.reader import make_reader

writer = rosbag2_py.SequentialWriter()
writer.open(rosbag2_py.StorageOptions(
    uri='my_bag',
    storage_id='mcap'
), rosbag2_py.ConverterOptions('', ''))

topic = rosbag2_py.TopicMetadata(
    id=0,
    name='test_topic',
    type='test_pkg/msg/McapGh1337',
    serialization_format='cdr')
writer.create_topic(topic)

for i in range(10):
    msg = McapGh1337()
    msg.header.stamp.sec = i
    msg.joint_health_flag = array('b', [i])
    writer.write(
                'test_topic',
                serialize_message(msg),
                i * 1_000_000_000)

del writer

with open("my_bag/my_bag_0.mcap", "rb") as f:
    reader = make_reader(f, decoder_factories=[DecoderFactory()])
    for schema, channel, message, ros_msg in reader.iter_decoded_messages():
        print(f"{channel.topic} {schema.name} [{message.log_time}]: {ros_msg}")
```

```
test_topic test_pkg/msg/McapGh1337 [0]: McapGh1337(header=Header(stamp=Time(sec=0, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[0])
test_topic test_pkg/msg/McapGh1337 [1000000000]: McapGh1337(header=Header(stamp=Time(sec=1, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[1])
test_topic test_pkg/msg/McapGh1337 [2000000000]: McapGh1337(header=Header(stamp=Time(sec=2, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[2])
test_topic test_pkg/msg/McapGh1337 [3000000000]: McapGh1337(header=Header(stamp=Time(sec=3, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[3])
test_topic test_pkg/msg/McapGh1337 [4000000000]: McapGh1337(header=Header(stamp=Time(sec=4, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[4])
test_topic test_pkg/msg/McapGh1337 [5000000000]: McapGh1337(header=Header(stamp=Time(sec=5, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[5])
test_topic test_pkg/msg/McapGh1337 [6000000000]: McapGh1337(header=Header(stamp=Time(sec=6, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[6])
test_topic test_pkg/msg/McapGh1337 [7000000000]: McapGh1337(header=Header(stamp=Time(sec=7, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[7])
test_topic test_pkg/msg/McapGh1337 [8000000000]: McapGh1337(header=Header(stamp=Time(sec=8, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[8])
test_topic test_pkg/msg/McapGh1337 [9000000000]: McapGh1337(header=Header(stamp=Time(sec=9, nanosec=0), frame_id=), solver_status=0, solver_obj=0.0, joint_health_flag=[9])
```

closed 2025-02-25 by jtbandes

kwesiRutledge-aescape, 2025-03-05:
Issue was reproduced earlier this week by other members of the team. This time, the issue when using the Foxglove Data API (which internally relies on this library).