foxglove/mcap #1348: Rust: WriterOptions cannot be used

Charlie-AuZone, 2025-03-24:
**The WriterOptions cannot be used**

- Version: 0.15.1
- Platform: Ubuntu 20.04

**Steps To Reproduce**
```rust
mcap::Writer::with_options(buf, mcap::WriterOptions::default());
```
This cannot be called because the function is not `pub`. This means that the user cannot configure write options beyond the default, such as changing the compression level between zstd and uncompressed.
https://github.com/foxglove/mcap/blob/main/rust/src/write.rs#L397

**Expected Behavior**
The `mcap::Writer::with_options` is `pub` and can be called by the user

bennetthardwick, 2025-03-24:
You can currently do `WriteOptions::default().create(..)` which will call `with_options` under the hood. I don't think is super intuitive though so I agree that `with_options` should be made public.

Charlie-AuZone, 2025-03-24:
Oh I didn't see that function, since the `mcap::Writer::new()` as just called `Self::with_options(writer, WriteOptions::default())` inside, I didn't find that. Thank you! I will edit the original message to include this information

PR #1351 (2025-03-26): rust: make Wrtier::with_options pub
### Changelog

Make `Writher::with_options` public

### Docs

None

### Description

Makes the `Writer::with_options` method in the Rust library public to easier support setting options.

This also bumps the version to 0.16.0 in preparation to release this and the breaking changes from #1341.

Fixes #1348.
commit b80d5bd2 (2025-03-26): rust: make Wrtier::with_options pub (#1351)

### Changelog

Make `Writher::with_options` public

### Docs

None

### Description

Makes the `Writer::with_options` method in the Rust library public to
easier support setting options.

This also bumps the version to 0.16.0 in preparation to release this and
the breaking changes from #1341.

Fixes #1348.
closed 2025-03-26 by bennetthardwick