foxglove/mcap #1353: mcap-cli go-module references incompatible mcap module

alokpr, 2025-04-02:
mcap-cli module references old mcap and ros module versions that are not compatible with the cli module.

So "go get github.com/foxglove/mcap/go/cli/mcap" indirectly pulls incompatible versions of dependencies that it does not build against.

https://github.com/foxglove/mcap/blob/main/go/cli/mcap/go.mod

bennetthardwick, 2025-04-02:
Hi @alokpr what is the reason behind doing `go get` on the CLI module?

We don't support installing the CLI via `go install`, etc. - if you want to build the CLI locally there are instruction here: https://mcap.dev/guides/cli#from-source.

Similar issue: https://github.com/foxglove/mcap/issues/1317

alokpr, 2025-04-02:
Ah, yes this is a duplicate of #1317.

I ran into this issue when importing mcap-cli into our build system, which has tooling to auto-generate BUCK targets using go module. I have a workaround for now by vendoring the compatible dependencies.

I am wondering if it would be possible to either update the versions in module file as part of the release process or just create a single top-level module instead of the workspace.

jtbandes, 2025-04-12:
> I am wondering if it would be possible to either update the versions in module file as part of the release process or just create a single top-level module instead of the workspace.

@bennetthardwick are you able to comment on this idea?

closed 2025-04-12 by jtbandes