foxglove/mcap #1400: [CPP]: mcap files written with 2.0.2 not compatible with 1.3.0, when message size exceed chunk size

facontidavide, 2025-06-15:
## Description
There is apparently a problem of back compatibility between MCAPs written with the latest version of MCAP (**2.0.2**) and the one found in ROS2 Jazzy, i.e. **1.3.0**.

For more details about how we found this issues, you can take a look at this issue in Cloudini:

https://github.com/facontidavide/cloudini/issues/15

Afters some debugging, I found that the error happens when the messages exceed the chunkSize.

A unit test and fix was introduced in 2.0.1
https://github.com/foxglove/mcap/blob/main/cpp/test/unit_tests.cpp#L681

But apparently the fix is incomplete, or at least not back compatible with 1.3.0.

### Expected Behavior
Writing a MCAP with messages that exceed the chunkSize using the latest MCAP 2.0.x should generate files that `ros2 bag play` can play correctly

### Actual Behavior

We get error message:
```
[rosbag2_storage_mcap]: cannot read MCAP in time order with no message indexes
```

## To Reproduce
Create a file as in the unit test mentioned above and try reading it with older version 1.3.0

## System (please complete the following information)
 - OS: [Ubuntu 24.04]
 - ROS 2 Distro: [Jazzy]
 - Install Method: [APT]
 - Version: [2.0.2]

## Additional context
Soooooooo...

I was doing some dubugging and I found this:

1. The very first message being written exceeds the chunkSize and this branch is taken:

[writer.inl#L573](https://github.com/foxglove/mcap/blob/main/cpp/mcap/include/mcap/writer.inl#L573)

2. But this line was never reached: https://github.com/foxglove/mcap/blob/main/cpp/mcap/include/mcap/writer.inl#L598

3. Therefore, this for loop is not executed

https://github.com/foxglove/mcap/blob/main/cpp/mcap/include/mcap/writer.inl#L784-L794

4. This means that the very first `chunkIndex` is never initialized

https://github.com/foxglove/mcap/blob/main/cpp/mcap/include/mcap/writer.inl#L779

james-rms, 2025-06-15:
is the ask to provide a 1.3.1 patch and backport it into jazzy?

defunctzombie, 2025-06-16:
@james-rms The way the issue is written makes it sound like v2 doesn't write mcaps correctly in this situation (saying the chunkindex is not initialized) which is what causes a older reader not to be able to read the file. Or is the claim that the mcap file is valid and the older reader has a bug?

facontidavide, 2025-06-16:
let me be more direct and less ambiguous. I don't believe this is an intended/correct behavior, but... you tell me.

[image]

james-rms, 2025-06-19:
@facontidavide I think the current 2.0.2 behavior is correct. My reading of your walkthrough is:
1. You start a new file
2. You write a channel and schema - these get written to the first chunk.
3. You write a very large message. The first chunk is closed immediately, with no messages. In this context, no message index records should be written after the chunk. A chunk index record is placed in the summary with message index length of zero.
4. Then, a second chunk is opened to write the new message into.

v1.3.0 has a bug where if the first chunk has no message indexes, it assumes all chunks have no message indexes, and it will refuse to read the MCAP in log-time order. This is either a bug or a limitation, depending on how you see it, but either way - it's no good for users of that library. I can either publish a v1.3.1 patch to cover this case, or if the ROS2 distribution versioning policy allows, update rosbag2_storage_mcap to v2.0.2. Either one would solve the problem. It's been a long time since I poked at this, so I'll see what the other maintainers think.

closed 2025-07-09 by james-rms